uavos/apx-v10-releases #29: Brakes and gimbal problem

smarttowel, 2019-06-12:
Gimbal can't moving, while ctr_brake=1. I think, bldc in gimbal should be ignore brakes command.

smarttowel, 2019-06-19:
Пишу за сутки.
Могу завтра организовать.

uavinda, 2019-06-19:
Ok. I'm available from 9am to 9pm Miami time.

uavinda, 2019-06-20:
The issue is related to custom BLDC firmware `gd3f28`. Can't find one in the APX or autopilot tree.

uavinda, 2019-06-20:
---------- Forwarded message ---------
From: Yury Kapacheuski <ky@uavos.com>
Date: Thu, Jun 20, 2019, 10:49
Subject: Обновление прошики BLDC
To: Pawel Mikhadzionak <pmv@uavos.com>, Aliaksei Stratsilatau <sa@uavos.com>,
Ruslan Lysenka <lr@uavos.com>, Egor Volkov <ve@uavos.com>

В данной версии система не выключает двигатели по команде “brakes” если в
конфигурации  не установлена настройка выхода на тормоз.
(FOC:Diskrets:dBrake)

closed 2019-06-20 by uavinda