uavos/apx-v10-releases #34: Too slow downloading nodes configuration via modem

smarttowel, 2019-06-19:
APX
modem: can't download nodes, multiple "Service timeout" in console
ghanta: works fine

GCU 9.x
modem: works fine
ghanta: works fine

smarttowel, 2019-06-19:
Firmware version 9.x

uavinda, 2019-06-19:
How is modem connected to PC?

smarttowel, 2019-06-19:
USB-COM converter "МастерКит"

uavinda, 2019-06-19:
Couldn't you attach telemetry file?

smarttowel, 2019-06-19:
I can, but tomorrow.
What kind of telemetry you need? Just ~1 minute, when I try to download configuration?

uavinda, 2019-06-19:
Correct.

smarttowel, 2019-06-20:
[2019_06_20_11_11_54_984-Heli_E_#2.telemetry.zip](https://github.com/uavos/apx-releases/files/3309433/2019_06_20_11_11_54_984-Heli_E_.2.telemetry.zip)

smarttowel, 2019-06-20:
My custom plugin set vars VM1 and VM2 every 10 msecs via setValue(...) method. 
By design VM* variables shouldn't send over network, but in real it's does.
I solved this problem by using setValueLocal(...) method.

closed 2019-06-20 by smarttowel

uavinda, 2019-06-20:
By design VM* variables shouldn't send over network, but in real it's does.

Should this behavior be fixed? If so - create issue.

Regards,
Aliaksei Stratsilatau
http://www.uavos.com/