uavos/apx-v10-releases #39: Streaming plugin fails to play sample stream

uavinda, 2019-07-07:
Gstreamer 1.14.2

MacOS:
(<unknown>:76194): GStreamer-CRITICAL **: 11:33:44.591: gst_bin_add_many: assertion 'GST_IS_ELEMENT (element_1)' failed
[20190707 11:33:44.591 W] app - GstPlayer::onErrorOccured - Can't create pipeline
[20190707 11:33:49.591 W] app - GstPlayer::onErrorOccured - Connection timeout

smarttowel, 2019-07-07:
URI of stream?

uavinda, 2019-07-07:
rtspt://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov

uavinda, 2019-07-07:
rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov

smarttowel, 2019-07-07:
I need some logs
`cd /Library/Frameworks/GStreamer.framework/Commands/`
`./gst-inspect-1.0 > ~/log.txt`
`gst-inspect-1.0 --version`

Can you repeat this problem on Linux?

smarttowel, 2019-07-07:
I can't repeat this problem on last build from `master` even with missed system GStreamer (`sudo rm -rf /Library/Frameworks/GStreamer.framework/`).
I think, this problem related to a GStreamer deployment rules. This rules works only in `release` build. Are you sure, that you build gcs in release?
Also, can you provide some info about your build or, maybe, binaries?

uavinda, 2019-07-07:
I test debug build. But it is supposed to work too.

uavinda, 2019-07-07:
Also, make your pull request to git-repos branch for this issue. Make it to
work properly with debug build.

smarttowel, 2019-07-07:
> I test debug build. But it is supposed to work too.
> […](#)

ApxDeployLibs product works only in release mode.

uavinda, 2019-07-07:
The git-repos branch has the new upcoming project structure.

uavinda, 2019-07-07:
Why do you need to have gst libs in app bundle?

smarttowel, 2019-07-07:
Because we provide bundle! Bundle means all-in-one app.

uavinda, 2019-07-07:
Not in debug mode.

smarttowel, 2019-07-07:
I don't understand what you want.

uavinda, 2019-07-07:
To make it work with OS gst default installation in debug mode. Or to
understand the reason why gst must be fixed or copied to bundle to work
with the streaming plugin.

smarttowel, 2019-07-08:
Fixed

closed 2019-07-08 by smarttowel