uavos/apx-v10-releases #40: Streaming plugin UI lacks URI setup and is not user friendly

uavinda, 2019-07-07:
Please, add the following features to the `streaming` plugin UI:

- select type of stream from enum, when selected, the corresponding options facts become visible as in the old implementation:
  - `URI` - **default** - text input for custom gst URI, remember last choices, support `rtspt` schema to select transport;
  - `RTSP` - add option for tcp transport, text input for URI, default schema and port assumed, i.e. should work when typed as `192.168.8.8:3555/cam` or `rtspt://127.0.0.1` (tcp transport override, option tcp ignored) or `10.0.0.1`, etc;
  - `TCP` - IP and port;
  - `UDP` - port;
  - `LOCAL` - enum to select available local sources, scanned by gst and shown as descriptive text names;

Check for supported schemas, received from gst and add or remove from the above enum. 

Add any other options to schemas when there may be a case.

All options should remember last choices in the `Ground Control.ini` or even better - in the dedicated `streaming.json` file, as f.ex. `joystick` plugin does with default values present in the app resources or fact options.

Consider to add any specific schemas or options if you think it is worth of.  For example, you can add options:
- `IP Camera` - input of the IP address, then plugin automatically scans for known URI paths of IP cameras by making web requests to the IP and fills out the correct URI (read only or enum). Add all known cameras we use now. F.ex. you input `10.1.1.9`, and the plugin fills out the enum fact with, let's say `rtsp://10.1.1.9:1234/stream/cam0/stream1` and `rtspt://10.1.1.9:1234/stream/cam0/stream1` options. The progress of scanning should be shown by enum fact.
- `SCAN` - think about implementing a simple IP scanner to detect known IP cameras according to current IP address of local host and provide selections for `URI` option.

This would resolve most issues and optimize our work a lot.

smarttowel, 2019-07-09:
Why we need `TCP` and `UDP` as separate options? `RTSP` can stream any video over udp or tcp. Moreover, no cams, that stream video over pure `TCP` or `UDP`, because:
1. `TCP` is not standard, not documented streaming protocol. It's [legacy](http://gstreamer-devel.966125.n4.nabble.com/tcp-with-uridecodebin-playbin-td4666478.html) from ffmpeg.
2. `UDP` often means rtp stream.

Both of these protocols **doesn't have encoding description** (rtsp do that).
So, we have two protocols, that we can use only in debug purpose with hardcoded muxers, encoders and decoders. In addition, `RTSP` can do it without any pain.

So, I want to remove `UDP` and `TCP` options.

uavinda, 2019-07-09:
Make these two options work the same way as qgroundcontrol does.

uavinda, 2019-07-09:
The `streaming` plugin should support all features present in qgroundcontrol with improvements and extra features described in this issue.

smarttowel, 2019-07-09:
12 days ago qgroundcontrol implement h265 udp streaming. This feature missed in old streaming plugin. Now they have two hardcoded decoders for udp (_but rtsp can resolve it automatically_).
Should I backport this feature?

uavinda, 2019-07-09:
Unless there's no features downgrade in comparison to qgroundcontrol - you
can do whatever you want. But pay attention to make UI as simple as
possible.

The user should be able to connect a video source without knowledge of URI
format for the gst, but knowing the IP address only. Thus I consider the
options TCP and UDP to be more suitable for our users.

Regards,
Aliaksei Stratsilatau
http://www.uavos.com/

smarttowel, 2019-07-09:
In case of TCP and UDP user should be know codec.

uavinda, 2019-07-09:
My assumption is that there are some gimbals or payloads we are not really
familiar with. To setup these payloads - qgroundcontrol users just choose
UDP option and use default port. It works out-of-the-box. We need the same
simplicity.

Of course, the implementation of HTTP/IP scanners would make this situation
**much better**, and generally speaking, you can remove all options, but
keep just these two. The first - scans known URLs by IP and proposes URIs.
The second - scans network for available IPs, then runs the first one to
propose the valid URIs, or does some other probes.

I agree to remove TCP/UDP options if you think it is not really necessary
to have ones.

Regards,
Aliaksei Stratsilatau
http://www.uavos.com/

uavinda, 2019-07-09:
The issue description **is not the requirement**. Propose your own user-friendly approach.
For example - you may want to remove all options, but keep one text input for the gst URI. And add an action button `scan`, which fills out the URI combo with suggested/found URIs. The `scan` action could open another setup options dialog, like IP range, etc.

smarttowel, 2019-07-09:
Ok, I will add support for TCP and UDP streaming with ability to choose decoder type for some unknown payload, that we can use in future (maybe).

uavinda, 2019-07-09:
A good idea would be to add some virtual protocols, or the ability to build
any pipeline, or to set any options through URI text input for advanced
users. The documentation page is surely welcomed to be added to docs site.

closed 2019-07-21 by smarttowel