uavos/apx-v10-releases #41: UI improvements for Fact

smarttowel, 2019-07-09:
1. [Validators](https://doc.qt.io/archives/qt-4.8/texthandling.html) for Fact.
2. Editable ComboBox (Fact::Enum)

uavinda, 2019-07-09:
Please, add more description to the problem.

smarttowel, 2019-07-09:
For example, we have TextInput for ip:port. We can make regexp validator for this format, so user immediately can see, valid input or not.

uavinda, 2019-07-09:
I think we should declare a new QString property for the Fact: `validator`. Then we must update qml FactButton editors, wherever they use TextInput - to support `validator` regex **when present**. Can you make it and propose the changes?

uavinda, 2019-07-09:
Don't forget to update `FactObject.qml` to support `validator`.

uavinda, 2019-07-09:
I just have to merge `git-repos` to `master` ASAP to continue releases.

smarttowel, 2019-07-11:
I will do it, when I have free time.
You can prepare release without this feature.