uavos/apx-v10-releases #42: import configuration feature problem

carbonlof, 2019-07-10:
**Describe the bug**
After an attempt to import the configuration from the old configuration file, a message that the configuration file is locked appeared, however when the configuration file is imported in local mode, a message appeared that the configuration is loaded.

**Desktop (please complete the following information):**
 - OS: Distributor ID: Ubuntu
Description:    Ubuntu 18.04.1 LTS
Release:        18.04
Codename:       bionic

uavinda, 2019-07-10:
Pls, attach this configuration file.

carbonlof, 2019-07-11:
[DESCKY GLIDERlAPICHIOK.nodes.txt](https://github.com/uavos/apx-releases/files/3380868/DESCKY.GLIDERlAPICHIOK.nodes.txt)

uavinda, 2019-08-22:
Can not reproduce this issue. Please provide more details, like step-by-step description how to reproduce it.