uavos/apx-v10-releases #45: Fact::toJson doesn't return yourself

smarttowel, 2019-07-11:
I don't know, how this method should work, but it's strange behaviour:
`Fact *jsonFact = new Fact(nullptr, "Name", "Title", "Descr", Text);
    jsonFact->setValue("Value");
    qDebug() << jsonFact->toJson();`
will print:
`QJsonObject()`

Also, maybe, full marshalling should be implemented (with datatype, descr, etc.)?

uavinda, 2019-07-11:
Noted.

uavinda, 2019-08-22:
This conversion functions are implemented to support settings storage, when facts are used as value handlers for settings. Agree to rename these methods to `getValues`, `setValues` as it was done with QVariantMap format.

Regarding marshalling - what is the intention? Any use cases?

closed 2019-08-22 by uavinda

smarttowel, 2019-08-23:
I have two problems.
First case:
```
Fact p(nullptr, "parent", "Title", "Descr", Fact::Text);
p.setValue("trololo");
Fact c1(&p, "child1", "Title", "Descr", Fact::Text);
c1.setValue("text1");
Fact c2(&p, "child2", "Title", "Descr", Fact::Text);
c2.setValue("text2");
qDebug() << p.valuesToJson(); //QJsonObject({"child1":"text1","child2":"text2"})
```

Second case:
```
Fact p(nullptr, "Name", "Title", "Descr", Fact::Text);
p.setValue("trololo");
qDebug() << p.valuesToJson(); //QJsonObject()
```

In all cases I can't see value of fact, in which I call valuesToJson().
It's exactly related to QSettings.