uavos/apx-v10-releases #46: Problems with Map plugin backends

smarttowel, 2019-07-17:
I want to draw QGeoPolygon with a hole inside (related to #32). I can do it in Qt 5.13 release with a mapboxgl plugin (see [this](https://stackoverflow.com/questions/57054554/its-possible-to-draw-mappolygon-with-a-hole-on-map-in-qml))

But, for some reason, we have hardcoded `plugin: Plugin { name: "itemsoverlay" }` in GCS code.
If I change this line to `plugin: Plugin { name: "mapboxgl" }` it works fine.

uavinda, 2019-08-19:
This week.

uavinda, 2019-08-22:
Mapbox-gl works twice slower than `itemsoverlay`, at least with Qt5.13 QML bindings. It also requires access token which is not free. Right now we can't port everything to `mapboxgl` as it would require to implement it all in JavaScript and WebView coupled to QML and `facts` backend. Although, this could be our future step and anyway, it should be performed in separate plugin to keep current functionality untouched.

smarttowel, 2019-08-22:
I don't speak about porting everything to mapboxgl backend.
I mean, that we can choose mapboxgl plugin in GCS, but it's not works as expected (no holes in polygon).

Strictly speaking, MapPolygon::geoShape is not documented feature right now, because only mapboxgl support this. But! In next releases, Qt will be add this feature to other backends and we have problems with that.

uavinda, 2019-08-22:
Then its gonna work with itemsoverlay as well, isn't it?

uavinda, 2019-08-22:
The reason why the [`itemsoverlay`](https://doc-snapshots.qt.io/qt5-dev/location-plugin-itemsoverlay.html) is used above `provider plugin` is that it makes it independent from underlying backend like `osm` or `mapbox-gl` or whatever. Strictly speaking, when you add an item to `mapbox-gl` map, it does not draw it natively through mapbox engine, but creates wrapped QML MapItem to draw it in the GL context of mapbox-gl which is relatively slow and 'non-native' way of doing things. I've made a test - 1000 waypoints are much slower with `mapbox-gl`. Sure, this will be improved in future versions of Qt.

To resolve your issue - I have added `ui.mapbase` reference to the JS context, so that your plugin could add items to the underlying Map, or control its parameters. Just change parent to be `ui.mapbase` not `map`.

In future - this would be very attractive to implement our flight planner natively in `mapbox-gl` JS with backend synchronization. Such things are either hard to implement and maintain in C++. Also, porting the application to different platforms (including Web) would be much easier.

smarttowel, 2019-08-23:
I just replace ui.map on ui.mapbase and I see errors, that ui.mapbase is undefined.

smarttowel, 2019-08-29:
Still not working

closed 2019-09-08 by uavinda