uavos/apx-v10-releases #51: Telemetry plugin sometimes can't show one variable

smarttowel, 2019-08-12:
I want to see one variable from my telemetry, so I do:

1. Click on green button (load latest telemetry)
2. Click on CLR button
3. Choose one variable (f.ex. user2) and no graphics displayed
4. Choose one more variable (f.ex., user3), or scale graphics and I see missed var.
[image]
[image]

uavinda, 2019-08-20:
This might be QWT issue. What happens when you just adjust zoom a little bit before your step 4?

smarttowel, 2019-08-20:
Yes, I can see missed var in this case.

smarttowel, 2019-08-20:
If it is QWT bug, what about Qt Charts usage?

uavinda, 2019-08-20:
I actually wanna move to QtCharts, but have lots of higher priority tasks..

closed 2019-08-20 by uavinda