uavos/apx-v10-releases #52: Videoplugin is broken in v10.1.154 release

smarttowel, 2019-08-13:
Problem occured only on Linux. Kubuntu 18.04.

But if I build project from sources (release and debug builds), plugin works fine.
I think, it's can be travis related problem. Maybe, we should make linux build under bionic, not xenial distro?

[log.txt](https://github.com/uavos/apx-releases/files/3496340/log.txt)

smarttowel, 2019-08-23:
Ok, now we don't have errors in console, but video still not working in build from travis.
With `GST_DEBUG="*:3"`:

Missing parser: H.264 (Main Profile) (video/x-h264, stream-format=(string)avc, alignment=(string)au, level=(string)3, profile=(string)main, codec_data=(buffer)014d401effe000, width=(int)1920, height=(int)1072, framerate=(fraction)113/4, pixel-aspect-ratio=(fraction)1/1, parsed=(boolean)true)

Looks like h264parse element not found.

uavinda, 2019-09-06:
Travis build log: https://api.travis-ci.com/v3/job/230905203/log.txt?log.token=2s8B8-hGqXuzrFRuNm0Omw

uavinda, 2019-09-08:
It looks like some libs are missing:
[image]

smarttowel, 2019-09-08:
Are you sure that it's screenshot from Release .AppImage build?
I check v10.1.190 right now and all libraries presented in .AppImage.

uavinda, 2019-09-09:
Sure I am. After copying of the missing libs, the image works fine.

closed 2019-09-09 by uavinda