uavos/apx-v10-releases #75: AppSettingFact default value and setMin call: non-evident behaviour

smarttowel, 2019-10-07:
```
f_scale = new AppSettingFact(settings, this, "scale", tr("scale"), "", Float, 0.1);
f_scale->setMin(0.01);
f_scale->setMax(1.0);
AppSettingFact::loadSettings(this);
qDebug() << f_scale->value(); //0.01
```

But expected, that f_scale should be equal to 0.1 (no value in config).
After some search, I found, that at line:
`
f_scale->setMin(0.01);`
we have f_scale == 0. So, 0 < 0.01, `setValue(0.01)` called. This call overrided in AppSettingFact. 0.01 implicitly written to config and we read it at AppSettingFact::loadSettings(this) call.

uavinda, 2019-10-09:
I plan to add ability to any `Fact` to permanently store its value in central app settings file. Thus, AppSettings class will be removed soon.

To resolve your specific issue - just call AppSettingsFact::load() method before setting the limits. It's not right, but to be refactored.

smarttowel, 2019-10-09:
In current implementation, `Fact::setValue(defaultValule)` in AppSettingFact constructor can be a not bad fix.

uavinda, 2019-11-01:
AppSettingFact  deprecated

closed 2019-11-01 by uavinda